pihomeserver/Kupiki-Hotspot-Script #172: self registered users cannot be listed/monitored/deleted

mickrow, 2019-06-26:
Standard settings allow new users to sign up via the portal page.

Those users however are not shown within users list in daloradius after that!
So there is no chance to know how many users have registered, impossible to restrict/disable/delete them.
Having the option to add more fields to it - like user's email, residence address, whatever your local law requires would be nice too. daloradius user info has a lot of fields for it ;)

kwaknet01, 2019-06-26:
[image] 2 users. registered via portal.

[image] list of users empty.

pihomeserver, 2019-06-26:
You can give a try to this tool to manage users
Else i will check where Daloradius is reading the users list. I guess that a record is missing to link registered users and the radinfo records

Photonoxx, 2019-08-29:
Hi, 
Any news about this ? 
Indeed it is possible to edit a self registered user, but you have to know the account name. By the way it is possible to found account name with the accounting tabs and viewing all records, but it's not optimum, even possible if many users use the hotspot.

pihomeserver, 2019-08-29:
Nope sorry. I miss time and i have no support from the community

Photonoxx, 2019-08-30:
Hi, 

I just check a bit the radius mysql database in Libreoffice (not a command line master), and I could find self registered user only in one table "radcheck" where password are stored, but not it two other ones "userinfo" and "userbillinfo" where some other infos appears about users.

I notice the self registered user only could have cleartext password, is this intended or could it be modified by default (I'm not supposed to know my user password).

Thanks anyway for your great and useful work and I hope community could be more supporting in the future