pihomeserver/Kupiki-Hotspot-Script #175: help connection dhcpcd lost

kwaknet01, 2019-08-07:
all is working after the installation of the script tried the portal and everything runs smoothly.  then suddenly wifi not working. i've tried rebooting still does not work.

kwaknet01, 2019-08-08:
[image]

i think raspberrypi-ui-mods is messing things up with network interfaces. cuz i've tried to install a fresh img. without installing gui now its running good so far. any advice on how to put gui without messing settings of kupiki?

pihomeserver, 2019-08-08:
Sorry the script has never been tested with raspbian. It’s only configured for raspbian lite. Help is welcome to make it compatible with the classic raspbian