pihomeserver/Kupiki-Hotspot-Script #176: Cant install, ends at this script is used to get installed on Raspbian Stretch Lite.

unmaskedd, 2019-08-10:
### Actual behavior

stop after checking that HOTSPOT NETWORK is not the same than the WAN_INTERFACE

### Expected behavior

Should continue the installation

#### Configuration parameters (if changed in the script)

Changed the name.

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : Raspberry pi 3b
- Operating system: Raspbian Buster Lite

### Additional comments

{Please write here, if there is something more to tell}

pihomeserver, 2019-08-11:
Sorry i don’t understand what the error is. No log, no message, no settings ...

closed 2019-08-26 by pihomeserver

vaaski, 2019-09-23:
This is raspbian Buster, the newest version.
I think your script refuses to run on it.

joeradie, 2019-10-09:
Yes, thats the problem (Raspbian Stretch Lite is now Raspbian Buster Lite). It's solved temporarily by commenting out the lines 468-474 of the pihotspot.sh, but I don't think that this solution is the best and safest one.
@pihomeserver could you please change this part of the installation script, that it works? Thx :)

pihomeserver, 2019-10-09:
Hi
Sure. Can you please display the result of the command from line 468 ? The one between comas

pihomeserver, 2019-10-12:
Script updated