pihomeserver/Kupiki-Hotspot-Script #177: help portal i can't execute php script inside index.html

kwaknet01, 2019-08-20:
i have python script to read the rfid card uid  
`#main.py
#!/usr/bin/python
import time
import RPi.GPIO as GPIO
import MFRC522
import os, sys
import string
from random import *

# Create an object of the class MFRC522
GPIO.setwarnings(False)
MIFAREReader = MFRC522.MFRC522()
#form = cgi.FieldStorage()

# Welcome message
#print("Tap Card Please")
#print("---------------")

# This loop checks for chips. If one is near it will get the UID
try:

  while True:

    # Scan for cards
    (status,TagType) = MIFAREReader.MFRC522_Request(MIFAREReader.PICC_REQIDL)

    # Get the UID of the card
    (status,uid) = MIFAREReader.MFRC522_Anticoll()

    # If we have the UID, continue
    if status == MIFAREReader.MI_OK:

        #print("Card UID: "+str(uid[0])+","+str(uid[1])+","+str(uid[2])+","+str(uid[3]))
        uidstr = '{}{}{}{}'.format(uid[0], uid[1], uid[2], uid[3])
        print(uidstr)
`

php script execute python 
`<html>
<body>
<?php
        $command = escapeshellcmd('sudo /usr/bin/python /home/pi/MFRC522-python/main.py');
        $output = shell_exec($command);
        echo "$output";
?>
</body>
</html>
`
[image]

the code works and prints the UID.

i want to use the UID of the RFID card use as  Username in index.html. 
i've tried but it doesnt work. ignore comment. 
[image]

pihomeserver, 2019-08-20:
Not sure that php5-fpm is installed and configured in the nginx configuration file to support PHP

kwaknet01, 2019-08-20:
how can i update nginx configuration file?

i also want to add my custom dashboard don't know how or where to edit nginx config file.

add to portal config.? is this correct? 
  location ~* \.php$ {
    fastcgi_pass unix:/run/php/php7.0-fpm.sock;
    include         fastcgi_params;
    fastcgi_param   SCRIPT_FILENAME    $document_root$fastcgi_script_name;
    fastcgi_param   SCRIPT_NAME        $fastcgi_script_name;
  }

pihomeserver, 2019-08-20:
Search google to add php support to nginx (not related to this project)

What is your custom dashboard ? Look at the configuration of NGINX and daloradius and adapt it for a new folder/url

kwaknet01, 2019-08-20:
thanks sir.