pihomeserver/Kupiki-Hotspot-Script #178: unable to find hotspot

taeralnar, 2019-08-21:
### Actual behavior

I executed the script but at the end it says ( unable to exceute last command ) & when i searched for hotspot that is supposed to be created by the Raspberry i didnt find anything ..... 
i wasnt able to access daloradius also ..... i dont know what is the problem ....
### Your environment

- Hardware : Raspberry Pi 3 Model B
- Operating system: Raspbian Stretch Lite

### Additional comments

sorry for inconvenience

pihomeserver, 2019-08-21:
Without logs or error messages it seems a little bit difficult to help ....

taeralnar, 2019-08-21:
i tried this command tail -f /var/log/pihotspot.log but it says no such file

pihomeserver, 2019-08-21:
Did you search in the folder ? What was displayed on the screen ? Did you reboot ? Did you check for different services like hostapd are up and running ?
Please keep in mind that we are far, with no access and have only words to talk !

taeralnar, 2019-08-21:
I reran the script hoping for solving the problem & i got these errors : 
::: configuring daloradius DB proxy configuration file 
sed : cant read check_returned_code : No such file or directory 
sed : cant read 0 : No such file or directory 
::: Reloading fail2ban local configuration 
ERROR could not find server 
::: Something went wrong with the last command . Please check the log file 

After Checking for hostapd service  it said : 
Hostapd.service 
Loaded : masked (/dev/null ; bad )
Active : inactive (dead) 

Services freeradius & chilli are active ..... 
Can these information help you to identify cause of the problem ?

taeralnar, 2019-08-22:
I googled alot for hostapd being not working & tried these commands : 
Sudo systemctl unmask hostapd 
Sudo systemctl enable hostapd 
Sudo systemctl start hostspd 
 

The hotspot then appeared but without password & when i enter any user name & password it stucks on this screen ?? 

https://ibb.co/Jy6JxSY

I also tried accesing daloradius again & it worked but even after adding a user & trying to enter them in the hotspot it also stucks on the same screen

pihomeserver, 2019-08-22:
Strange as this is a known bug since march (#163 ) and the bug fixing has been implented in the latest releases ...

Do you get an error message after a while in the portal webpage ? Try to connect from a laptop/pc and have a look at the developer console to see any error message

taeralnar, 2019-08-22:
Tried to connect to the portal from my ipad .... after entering user & pass it gives me blank white page with ( 405 timeout ) error !!!

pihomeserver, 2019-08-22:
Please stop services coova and freeradius then start both in debug mode. Then rery to login and check outputs to see if you reach coova and then if it tries to authenticate and authorize in freeradius
If coova is not reached, please provide parameters of the configuration for the LAN and WAN

taeralnar, 2019-08-22:
i left all parameters as default .... didnt change anything

i stopped both freeradius and chilli the ran the command sudo chilli --fg --debug and thats what i got : 

chilli[2100]: (Re)processing options [/var/run/chilli.2100.cfg.bin]
chilli[2101]: running chilli_opt on /var/run/chilli.2100.cfg.bin
chilli[2100]: PID 2100 rereading binary file /var/run/chilli.2100.cfg.bin
chilli[2100]: Loading modules
chilli[2100]: PID 2100 reloaded binary options file
chilli[2100]: CoovaChilli 1.4. Copyright 2002-2005 Mondru AB. Licensed under GPL. Copyright 2006-2012 David Bird (Coova Technologies). Licensed under GPL. See http://coova.github.io/ for details.
chilli[2100]: chilli_main(7416): clock realtime sec 1566489929 nsec 508658269
chilli[2100]: chilli_main(7423): clock monotonic sec 496 nsec 33921737
chilli[2100]: TX queue length set to 100
chilli[2100]: Running /etc/chilli/up.sh
chilli[2100]: Hashlog 8 253 256
chilli[2100]: Net SNDBUF 163840
chilli[2100]: Net RCVBUF 163840
chilli[2100]: device wlan0 ifindex 3
chilli[2100]: lo  address family: 17 (AF_PACKET)
chilli[2100]: eth0  address family: 17 (AF_PACKET)
chilli[2100]: wlan0  address family: 17 (AF_PACKET)
chilli[2100]: docker0  address family: 17 (AF_PACKET)
chilli[2100]: lo  address family: 2 (AF_INET)
chilli[2100]: eth0  address family: 2 (AF_INET)
chilli[2100]: tun0  address family: 2 (AF_INET)
chilli[2100]: docker0  address family: 2 (AF_INET)
chilli[2100]: tun1  address family: 2 (AF_INET)
chilli[2100]: dhcp_hashinit(490): hash table size 64 (56)
chilli[2100]: dhcp_sendGARP(338): GARP: Replying to broadcast
chilli[2100]: dhcp_new(1191): dhcpif (wlan0) IPv6 address ::
chilli[2100]: RADIUS client 0.0.0.0:3799
chilli[2100]: Address already in use: bind() failed!
chilli[2100]: Failed to create radius

-----------------------------------------------------------------------------------

then i ran the command sudo freeradius -x and thats what i got :+1: 

FreeRADIUS Version 3.0.12
Copyright (C) 1999-2016 The FreeRADIUS server project and contributors
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE
You may redistribute copies of FreeRADIUS under the terms of the
GNU General Public License
For more information about these matters, see the file named COPYRIGHT
Starting - reading configuration files ...
including dictionary file /usr/share/freeradius/dictionary
including dictionary file /usr/share/freeradius/dictionary.dhcp
including dictionary file /usr/share/freeradius/dictionary.vqp
including dictionary file /etc/freeradius/3.0/dictionary
including configuration file /etc/freeradius/3.0/radiusd.conf
including configuration file /etc/freeradius/3.0/proxy.conf
including configuration file /etc/freeradius/3.0/clients.conf
including files in directory /etc/freeradius/3.0/mods-enabled/
including configuration file /etc/freeradius/3.0/mods-enabled/replicate
including configuration file /etc/freeradius/3.0/mods-enabled/cache_eap
including configuration file /etc/freeradius/3.0/mods-enabled/dynamic_clients
including configuration file /etc/freeradius/3.0/mods-enabled/logintime
including configuration file /etc/freeradius/3.0/mods-enabled/realm
including configuration file /etc/freeradius/3.0/mods-enabled/utf8
including configuration file /etc/freeradius/3.0/mods-enabled/exec
including configuration file /etc/freeradius/3.0/mods-enabled/always
including configuration file /etc/freeradius/3.0/mods-enabled/soh
including configuration file /etc/freeradius/3.0/mods-enabled/sql
including configuration file /etc/freeradius/3.0/mods-config/sql/main/mysql/queries.conf
including configuration file /etc/freeradius/3.0/mods-enabled/attr_filter
including configuration file /etc/freeradius/3.0/mods-enabled/radutmp
including configuration file /etc/freeradius/3.0/mods-enabled/preprocess
including configuration file /etc/freeradius/3.0/mods-enabled/sradutmp
including configuration file /etc/freeradius/3.0/mods-enabled/digest
[... 1098 lines cut ...]
Ready to process requests
(4) Received Access-Request Id 204 from 127.0.0.1:49683 to 127.0.0.1:1812 length 308
Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator!  (Shared secret is incorrect.)
Waking up in 0.3 seconds.
(4) Cleaning up request packet ID 204 with timestamp +56
Ready to process requests
(5) Received Access-Request Id 204 from 127.0.0.1:49683 to 127.0.0.1:1812 length 308
Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator!  (Shared secret is incorrect.)
Waking up in 0.3 seconds.
(5) Cleaning up request packet ID 204 with timestamp +66
Ready to process requests
(6) Received Access-Request Id 204 from 127.0.0.1:49683 to 127.0.0.1:1812 length 308
Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator!  (Shared secret is incorrect.)
Waking up in 0.3 seconds.
(6) Cleaning up request packet ID 204 with timestamp +76
Ready to process requests
(7) Received Access-Request Id 204 from 127.0.0.1:49683 to 127.0.0.1:1812 length 308
Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator!  (Shared secret is incorrect.)
Waking up in 0.3 seconds.
(7) Cleaning up request packet ID 204 with timestamp +86
Ready to process requests

------------------------------------------------

sorry for inconvenience as iam just a newbie

pihomeserver, 2019-08-22:
Please run as root the command `ifconfig -a`

taeralnar, 2019-08-22:
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:65:a1:ee:c1  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.13  netmask 255.255.255.0  broadcast 192.168.1.255
        ether b8:27:eb:03:c1:e9  txqueuelen 1000  (Ethernet)
        RX packets 2912  bytes 139609 (136.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 108  bytes 14784 (14.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 19  bytes 2255 (2.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 19  bytes 2255 (2.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

tun0: flags=81<UP,POINTOPOINT,RUNNING>  mtu 1500
        inet 192.168.10.1  netmask 255.255.255.0  destination 192.168.10.1
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 100  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4419<UP,BROADCAST,RUNNING,PROMISC,MULTICAST>  mtu 1500
        ether b8:27:eb:56:94:bc  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 144 (144.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pihomeserver, 2019-08-22:
Also there is an issue with Coova as it says that address is already in use
Did you reboot ? If no please reboot and retry
If yes (or does not work after reboot) please do `ps -ef | grep chilli`and paste result here

taeralnar, 2019-08-22:
rebooted and tried debugging chilli again gave me this result : 
chilli[1694]: dhcp_receive_arp(6032): ARP: Did not ask for router address: 192.168.10.1
chilli[1694]: dhcp_receive_arp(6034): ARP: Asked for target: 192.168.10.188
chilli[1694]: dhcp_receive_ip(3760): Address found
chilli[1694]: dhcp_receive_arp(5954): ARP: 90-B6-86-69-C7-16 asking about 192.168.10.189
chilli[1694]: dhcp_receive_arp(6032): ARP: Did not ask for router address: 192.168.10.1
chilli[1694]: dhcp_receive_arp(6034): ARP: Asked for target: 192.168.10.189

taeralnar, 2019-08-22:
and thats the result of the this command ps -ef | grep chilli :+1: 

root      1690  1525  0 17:30 pts/0    00:00:00 sudo chilli --fg --debug
chilli    1694  1690  0 17:30 pts/0    00:00:00 chilli --fg --debug
chilli    1763  1694  0 17:31 pts/0    00:00:00 chilli --fg --debug
pi        1770  1525  0 17:33 pts/0    00:00:00 grep --color=auto chilli

pihomeserver, 2019-08-22:
chilli is running 2 times please kill all and start only one
Also are you sure that no hardware is on the 192.168.10.X ? message is strange from DHCP requests

taeralnar, 2019-08-22:
i rebooted the pi but once i run sudo chilli --fg --debug chilli runs twice .... i dont know whats the problem ..... 
how can i check if there is hardware on 192.168.10.X ??
can the problem be solved if i flashed anew img and tried changing the ip of raspberry ?

pihomeserver, 2019-08-23:
Because of the sudo that's normal (1 sudo, 1 command)
For the network you should know if you have other routers install at home/office that could potentially assign IP's on that address range

taeralnar, 2019-08-23:
YYYYYYes .......... i installed a fresh img of Raspbian Stretch Lite ..... changed ip address to 10.10.10.1 disabled registering users and installed Haserl and it finally worked ^_^ ^_^ 

it works perfect on android devices ..... tried on ipad and even if it gives blank white page with ( 405 not allowed error ) but if you typed correct user and pass it will connect to internet ....
The problem is with my laptop ...... when i connect it gives me limited connection and the captive portal doesnt appear !!! 
is there any solution for this ??

pihomeserver, 2019-08-23:
Please open a dedicated ticket for the new issue

closed 2019-08-23 by pihomeserver