pihomeserver/Kupiki-Hotspot-Script #179: Unable to reach portal on pc

taeralnar, 2019-08-23:
### Actual behavior

Hotspot works perfect on android devices but on my ipad even if i disabled registering users the portal still shows it ......
https://ibb.co/b3B8k1P

And when i enter user and password it shows me this screen ....

https://ibb.co/DkQ5JkY

Also on pc i gives me limited internet connection and i cant reach the portal at all 

### Expected behavior

{Please write here}

#### Configuration parameters (if changed in the script)

Raspberry ip : 10.10.10.1 
Network ip : 10.10.10.0 
Installed Haserl 
Disabled registering users 

#### Logs

```
// write log messages here, if possible
// like output of the script or system messages
// The more information you provide, the best answer you will get
```

### Your environment

- Hardware : Raspberry pi 3 model b
- Operating system: Raspbian Stretch Lite

### Additional comments

{Please write here, if there is something more to tell}

pihomeserver, 2019-08-26:
There is something wrong in the configuration as i see that the text for the checkbox for terms and conditions is not displayed
Please look at the log of nginx to see what the error is. 
Also that's pretty strange that the client OS is creating problems for the hotspot a sit's not related to ...

taeralnar, 2019-08-27:
this is log of nginx when i try to open the portal from my ipad :

10.10.10.2 - - [27/Aug/2019:13:19:55 +0100] "GET / HTTP/1.1" 200 1319 "-" "Mozilla/5.0 (iPad; CPU OS 9_3_5 like Mac OS X) AppleWebKit/601.1.46 (KHTML, like Gecko) Version/9.0 Mobile/13G36 Safari/601.1"

taeralnar, 2019-08-27:
PC problem was solved by running netsh in ip reset in cmd ...... 

But on my ipad still i cant login .... Tried changing browser or renewing ip but (405 Not Allowed ) error is still popping up whenever i try to login to the portal