pihomeserver/Kupiki-Hotspot-Script #181: In on eth0 and out on usb modem

raitpngman, 2019-10-04:
Just wanted to share my experience.
I wanted traffic to come in on eth0 and out to the internet on a usb modem that shows up as eth1. I did not want the wifi to be used.  I also have really flaky internet.
I started with raspian buster. Because of my low bandwidth, I manually installed all the prerequisites in the script, downloaded coova-chilli and compiled and installed it according to the script, and downloaded all the git and wget files to /usr/src and made sure the name matched the script. Then I commented out all the lines related to that. I commented out all the lines related to hostapd.
I ran into one bug with ssl_match_hostname not being found. It looks for it in /var/lib instead of /var/local/lib or vice versa. I manually copied it into the other folder and then the script ran all the way through. It is currently working with a few problems.
One of the problems is I can't get to the admin page. It seems to be listening on eth1 which is my usb modem to the outside. I need to find some way to change that. I'll keep you posted on how it goes.

raitpngman, 2019-10-04:
After running the setup script, daloradius is now listening on the usbmodem ip instead of the eth0 ip (because of my different setup above). How do I switch this?

pihomeserver, 2019-10-04:
Never tested but the issue could be from the nginx configuration that is listening on default interface on port 80. You should look at the file /etc/nginx/sites-available/default

raitpngman, 2019-10-04:
Thanks for the tip, but I couldn't figure it out. I changed the 80 to 70  in the default. Then tried [ip address]:70 but it didn't go anywhere, stuck on "connecting". [ip address]:80 still brings up the portal. I wondered if it was ip tables and added "ipt -A INPUT -p usp -s 10.10.104.1 --dport 70 -j ACCEPT" to the start of /etc/chilli/up.sh, but it didn't help. Any other ideas? I did install the alternative admin when I ran the script originally.

pihomeserver, 2019-10-04:
If the site still presents on 80 then the configuration has not been updated. You restart Nginx or reload it ? Also changing the port will not be enough. You have to expose it on another IP of the expected interface

raitpngman, 2019-10-04:
I changed the port to 70 in default, saved and rebooted. At [ip address]:80 the portal login still comes up. at [ip address]:70 it spins its wheels and nothing happens.
eth0 is my incoming traffic and my network. If I understand correctly, I'll have to assign a second ip address to that interface and then setup Ngnix to listen on that address to send to daloradius. I'll give it a shot tomorrow.

raitpngman, 2019-10-04:
I want to start the leases for the network at x.x.x.100, but I can't seem to find where that setting is. I've looked through all the dhcp files but can't find it. Can you point me in the right direction?

raitpngman, 2019-10-05:
> Never tested but the issue could be from the nginx configuration that is listening on default interface on port 80. You should look at the file /etc/nginx/sites-available/default

Did some more troubleshooting. in the /sites-available/default, it points to the daloradius install. So I removed the portal from the sites-enabled folder so it would run the settings in the default folder, but it came up with a 502 bad gateway error. Is the daloradius not configured right? Does this have anything to do with installing the kupiki admin interface?

pihomeserver, 2019-10-11:
Daloradius must be available on the interface connected on the internet, not the one exposed. So create a nginx file for it and in it specify the IP adress
That should do it