pihomeserver/Kupiki-Hotspot-Script #183: Erreur lors de l'installation | Error during installation

m290591, 2019-10-26:
Bonjour,
je souhaite installer votre projet sur mon Raspberry Pi 4 sous raspbian Buster. Voici la première erreur lors de l'installation.

`
::: Build the Docker image of Portal backend
ImportError: No module named ssl_match_hostname
`

Possible de contourner le problème en exécutant cette commande avant de relancer le script d'installation.
`cp -r /usr/local/lib/python2.7/dist-packages/backports/ssl_match_hostname/ /usr/lib/python2.7/dist-packages/backports`

Installation complète !

Lors d'une tentative d'ouverture de la page de connexion de daloradius j'obtiens une erreur 502.

Je m’aperçois que la configuration nginx pose problème (default et portal) je passe de php7.0 à 7.3
`fastcgi_pass unix:/var/run/php/php7.3-fpm.sock;`

L'erreur 502 lors de l'ouverture de la page daloradius disparait.

Mais un dernier problème persiste... J'arrive à créer un utilisateur depuis la page de connexion du portail (hotspot), mais impossible de se connecter ! Lors d'une tentative de connexion une animation de chargement apparait puis rien...

Des tentatives de connexion apparaissent dans les logs, mais le portail bloque.

Pouvez-vous m'aider ?

Merci

____________________________________________________________________

Hello,
I want to install your project on my Raspberry Pi 4 under raspbian Buster. Here is the first error during the installation.

`
::: Build the Docker image of Portal backend
ImportError: No module named ssl_match_hostname
`

You can work around this problem by running this command before restarting the installation script.
`cp -r /usr/local/lib/python2.7/dist-packages/backports/ssl_match_hostname/ / usr / lib / python2.7 / dist-packages / backports`

Complete installation!

When trying to open the daloradius login page I get a 502 error.

I realize that the nginx configuration is problematic (default and portal) I go from php7.0 to 7.3
`fastcgi_pass unix: /var/run/php/php7.3-fpm.sock;`

Error 502 when opening the daloradius page disappears.

But one last problem persists ... I can create a user from the portal login page (hotspot), but can not connect! When trying to connect a loading animation appears then nothing ...

Login attempts appear in the logs, but the portal blocks.

Can you help me ?

Thank you

pihomeserver, 2019-10-27:
Thank you for the feedback. A new release should be available soon with most of the bug fixes

About the new user unable to connect, in order to debug, can you please stop chilli and freeradius services then start each in a new terminal in debug mode
When you will try to connect you should see error messages

m290591, 2019-10-28:
Merci pour votre réponse, les problèmes sont résolu après plusieurs tentatives de réinstallation depuis zero. 

Voici la méthode la plus simple avant de démarrer le script kupiki il faut exécuter cette commande
`apt-get install -y python-backports.ssl-match-hostname`

Puis de passer les version de php de 7.0 à 7.3 dans la configuration de Nginx pour default et portal
`fastcgi_pass unix: /var/run/php/php7.3-fpm.sock; `

Maintenant le hotspot fonctionne correctement ! Merci pour ce super script. 🥳😎

___________________

Thanks for your reply, the problems are solved after several attempts to reinstall from scratch.

Here is the simplest method before starting the kupiki script
`apt-get install -y python-backports.ssl-match-hostname`

Then upgrade to the PHP version from 7.0 to 7.3 in the Nginx configuration for default and portal
`fastcgi_pass unix: /var/run/php/php7.3-fpm.sock; `

Now the hotspot is working properly! Thank you for this great script. 🥳😎