pihomeserver/Kupiki-Hotspot-Script #184: Mac address authentication

demperor13, 2019-10-29:
Excuse me ..... is there any way to add a specific mac so that it bypasses login page and directly access the internet ..... 

i tried mac address authentication option in daloradius but it requires you to enter the mac address as a username ....

### Your environment

- Hardware : Raspberry pi 3
- Operating system: Raspbian Stretch lite

### Additional comments

{Please write here, if there is something more to tell}

pihomeserver, 2019-10-31:
You have to activate the option in Coova in order to push the mac address to Freeradius as the username

demperor13, 2019-10-31:
how can i do this ? 
Sorry for inconvinence but i am just a beginner

pihomeserver, 2019-11-01:
In coova configuration file add `/etc/chilli/config`
```
HS_MACAUTH=on
HS_MACPASSWD=<what ever you want, but same than the one you set when you created the user>
```
So you have to create users with the same password for all

demperor13, 2019-11-01:
Tried this method and still portal webpage is popping up whenever i tries to access the network .....

Searched in google for a while and found a solution by editing /etc/chilli/defaults file and adding the mac address i want and finally it worked ....

Thanks alot for this great project and for your rapid response to any issues

closed 2019-11-16 by pihomeserver