pihomeserver/Kupiki-Hotspot-Script #185: Not able to access the admin.

uplsourav, 2019-11-15:
### Actual behavior

{just installed kupiki hotspot with admin enabled, able to see the hotspot, after connecting to the hotspot cp is opening}

### Expected behavior

{Ope the admin  with the URL, which I get at the end of the execution of the script }

#### Configuration parameters (if changed in the script)

{change the path of docker composer. /usr/local/bin/docker-compose to /usr/bin/docker-compose}

### Your environment

- Hardware : {raspberry pi 3 b+}
- Operating system: {Raspbian Buster Lite
Minimal image based on Debian Buster
Version: September 2019
Release date:2019-09-26
Kernel version:4.19
Size:435 MB}

Want to access the admin. Kindly help.

0x290591, 2019-11-15:
Hi,
try http://{ YOUR - IP}:8080

Maksymilian

uplsourav, 2019-11-16:
@0x290591 Thank you it's worked. now I'm able to see the admin page.

But not able to log in. whenever trying to login it's saying incorrect username password.
So I'm closing this issue and opening a new issue.

closed 2019-11-16 by uplsourav