pihomeserver/Kupiki-Hotspot-Script #188: error help.

kwaknet01, 2019-11-17:
[image]

[image]

is there something with the updated script? i can't seem to complete installation stuck here. tried reflashing and run installation script several times already.

pihomeserver, 2019-11-17:
I don't know. What pi ? what os ? what in the log ? etc.

kwaknet01, 2019-11-21:
im using raspbian stretch lite.. raspberry 3 b+,  i have'nt replicate the above error, but now i'm having issues with docker-compose not found. -su : /usr/local/bin/docker-compose: No such file or directory. still i have no success in building image.

kwaknet01, 2019-11-21:
how can i install docker-compose before running the script? i've notice it doesn't seem to run the command in the script for installing docker compose..

pihomeserver, 2019-11-21:
what is the result of `type docker-compose` ? if an error then install it using apt

kwaknet01, 2019-11-23:
error on script
`-bash: /usr/local/bin/docker-compose: No such file or directory`

but it is install on /usr/bin/docker-compose
`which docker-compose`
`/usr/bin/docker-compose`

kwaknet01, 2019-11-23:
`docker-compose version`
`docker-compose version 1.21.0, build unknown`
`docker-py version: 3.4.1`
`CPython version: 3.7.3`
`OpenSSL version: OpenSSL 1.1.1d  10 Sep 2019`

pihomeserver, 2019-11-24:
Please add a link from /usr/bin version to /usr/local/bin or update all calls in the script

I have to update the script to make the location dynamic