pihomeserver/Kupiki-Hotspot-Script #189: Error during installation on dietpi :

LoloPicolo, 2019-11-17:
### Actual behavior
I would like to install your project but it fails
current configuration is 

### Expected behavior
I would like to finalize install

#### Configuration parameters (if changed in the script)
no change
#### Logs

::: Checking for tun module
::: Insert tun module if existing (for Raspbian Jessie Lite)
::: Modprobe module (no check - useless if already loaded)
::: Checking for tun module
tun                    45056  0
::: Checking if LAN interface already exists
::: LAN interface not found. Upgrading the system first
::: Upgrading the distro. Be patient

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Lecture des listes de paquets…
Construction de l'arbre des dépendances…
Lecture des informations d'état…
Calcul de la mise à jour…
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
::: Upgrading the distro. Be patient
::: Please reboot and run the script again

### Your environment

- Hardware : Raspberry Pi 3 Model B (armv71)
- Operating system: DietPi latest version 6.26.3
- nginx, mariadb, hostap installed

### Additional comments
Thanks for your work. if i can help to install it on DietPi and support,

pihomeserver, 2019-11-19:
I don't understand what the problem is. The script displays a request for a reboot and a restart of the script. You did it ?

LoloPicolo, 2019-11-19:
It seems that for DietPi some packages are missing. i use the following before use your script and i move forward on installation
Before install your script on a DietPi, you need to execute the following
apt-get install collectd
apt-get install net-tools

currently i need to understand why isc-dhcp-server and proftp services are not working anymore.

pihomeserver, 2019-11-20:
Collectd is already includedin the list. I can add net-tools in case of if needed

yelooooo422, 2023-08-19:
::: Activating the LAN interface
:::
::: Something went wrong with the last command. Please check the log file

Can someone help me with this? I've been trying to look for a solution and reinstall the script but I still got the same error.