pihomeserver/Kupiki-Hotspot-Script #190: Docker-compose

OrbisTerrarumProLiberi, 2019-11-30:
### Actual behavior

Error install
### Expected behavior

Installation completed.

#### Configuration parameters (if changed in the script)

none
#### Logs

```
::: Build the Docker image of Portal backend
-bash: /usr/local/bin/docker-compose: No such file or directory
:::
::: Something went wrong with the last command. Please check the log file

### Your environment

- Hardware : Raspberry Pi 3
- Operating system: Rasbian Buster Desktop

### Additional comments

Any help will be much appreciated. Thanks

pihomeserver, 2019-11-30:
Quick search in the issues list : #190

OrbisTerrarumProLiberi, 2019-11-30:
fixed by eding script.
su - kupiki -c "cd /home/kupiki/kupiki-portal-backend && /usr/local/bin/docker-compose build"
to
su - kupiki -c "cd /home/kupiki/kupiki-portal-backend && /usr/bin/docker-compose build"

must have not noticed it earlier. Thanks!

closed 2019-11-30 by OrbisTerrarumProLiberi