pihomeserver/Kupiki-Hotspot-Script #191: Cant access daloradius

OrbisTerrarumProLiberi, 2019-11-30:
### Actual behavior

<ipaddress>/daloradius gives 404 Not Found
<ipaddress> 502 Bad Gateway

### Expected behavior

Daloradius web interface 

#### Configuration parameters (if changed in the script)

{Please write here}

#### Logs

Installation was successful

### Your environment

- Hardware : Raspberry Pi 3
- Operating system: Raspbian lite

### Additional comments

Any help will much appreciated. Thanks!

pihomeserver, 2019-11-30:
Check nginx logs in order to have more details and crosscheck with its configuration

OrbisTerrarumProLiberi, 2019-12-01:
updated php.ini
added: cgi.fix_pathinfo=0

changed /etc/nginx/sites-available
from
 fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
to
 fastcgi_pass unix:/var/run/php/php7.3-fpm.sock

closed 2019-12-01 by OrbisTerrarumProLiberi