pihomeserver/Kupiki-Hotspot-Script #194: Rpi2 compatible?

gibethub, 2020-01-03:
Update rights for nginx sites
: 
Build the docker image of portal backend
: 
Something went wrong with the last command...
Rpi2 with rtl8192cu and rt2800usb
Nothing modified with the original script...

dev1098, 2020-01-04:
My initial install failed there too.    Rpi 3b+, Buster.

A symlink fixed that issue:

`# ln -s /usr/bin/docker-compose /usr/local/bin/docker-compose`

So now I have a captive portal, but cannot login to user management -- Nginx throws a 502 Bad Gateway.    I ran the script a second time (yes, bad me...) so I'm going to start over and make sure to create the link first.

Many thanks @pihomeserver.   This has wildly exceeded my expectations for a guest hotspot.

gibethub, 2020-01-05:
symlink works, thanks but other problem occurs...
Created symlink /etc/systemd/system/multi-user.target.wants/freeradius.service → /etc/systemd/system/freeradius.service.
::: Creating fail2ban local configuration
::: Configuring fail2ban jail rules
::: Reloading fail2ban local configuration
OK
::: Disable fail2ban service
Synchronizing state of fail2ban.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable fail2ban
Removed /etc/systemd/system/multi-user.target.wants/fail2ban.service.
::: Stopping fail2ban service
::: Create banner on login
::: Append script version to the banner
::: Changing banner rights
::: Activating the banner for SSH
::: 
::: Reload configuration for SSH service
::: 
::: Something went wrong with the last command. Please check the log file
:::

pihomeserver, 2020-01-05:
Look at the log and paste the error
Else try/stop start the ssh service. That should not be a blocking issue

gibethub, 2020-01-05:
Reload configuration for SSH service
ssh.service is not active, cannot reload

pihomeserver, 2020-01-05:
So activate it and start it. I guess you don't connect to the pi via ssh ...

dev1098, 2020-01-05:
I tracked down the 502 error when accessing daloRADIUS:

`# tail /var/log/nginx/error.log`
.....
`2020/01/04 19:36:17 [crit] 12107#12107: *1 connect() to unix:/var/run/php/php7.0-fpm.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.x.x, server: _, request: "GET / HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock:", host: "192.168.x.x"`

Fixed by pointing the fastcgi_pass location to:  `/var/run/php/php7.3-fpm.sock`

in both files located in `/etc/nginx/sites-available`

Everything is working!

gibethub, 2020-01-05:
I will try a clean install... Reformatting sd card...

gibethub, 2020-01-06:
"": Installing the portal backend in the user kupiki home folder
::: Updating Captive Portal backend configuration file
::: Update rights for NGINX sites
::: 
::: Build the Docker image of Portal backend
::: Adding portal backend in systemd startup
::: Disabling Portal backend service
::: Activating Portal backend service
::: Starting Kupiki Portal Backend service
::: Activating registration in the Portal site
::: Adding Freeradius in systemd startup
::: Activating Freeradius service
::: Creating fail2ban local configuration
::: Configuring fail2ban jail rules
::: Reloading fail2ban local configuration
::: Disable fail2ban service
::: Stopping fail2ban service
::: Create banner on login
::: Append script version to the banner
::: Changing banner rights
::: Activating the banner for SSH
::: 
::: Reload configuration for SSH service
ssh.service is not active, cannot reload.""
a clean automatic script (just few replacement e.g. hotspot name and pass plus the symlink for docker-compose)
rpi 2 / 2.5a power adp / 16 gb strontium
raspian buster full
lg tv/wireless mouse
eth (source of internet)
r5370 usb wifi

thanks
(newbie here)

dev1098, 2020-01-07:
`# sudo systemctl enable ssh`

`# sudo systemctl start ssh`

gibethub, 2020-01-07:
Creating fail2ban local configuration
Configuring fail2ban jail rules
Reloading fail2ban local configuration
Could not find server...

dev1098, 2020-01-08:
Difficult.   It might be related to your Python version:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=496633

What do you get from:

`$ python -V`

Mine is 2.7.16

gibethub, 2020-02-04:
> Difficult. It might be related to your Python version:
> 
> https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=496633
> 
> What do you get from:
> 
> `$ python -V`
> 
> Mine is 2.7.16

Python 2.7.16

jcartner-young, 2020-07-03:
Any further input on compatibility with RPi2B?  I ran into multiple errors during and after script execution.  I have logged an issue.