pihomeserver/Kupiki-Hotspot-Script #195: "Something went wrong..."

stewert69, 2020-01-28:
https://github.com/pihomeserver/Kupiki-Hotspot-Script/blob/8acbc8b14df962d8b031ee7c7fc85dc8da183bb8/pihotspot.sh#L1048

After following all instructions, using Raspbian Buster, Raspbian Buster Lite and Raspbian Jessie Minibain, I always get an error in the script at line 1048 where it tries to "Build the Docker image of Portal backend". The result is " -bash: /usr/local/bin/docker-composer: No such file or directory"
Any guidance in this matter would be greatly appreciated.

pihomeserver, 2020-01-28:
Look at other issues. That's a known point where the docker-compose is newly installed in another folder. You have to change it manually or better create a pull request with updates

stewert69, 2020-01-28:
> Look at other issues. That's a known point where the docker-compose is newly installed in another folder. You have to change it manually or better create a pull request with updates

What folder should it be installed in then?

pihomeserver, 2020-01-28:
Try to look at the new location with the type command. Not sure about the location in latest debian release