pihomeserver/Kupiki-Hotspot-Script #199: Stuck at loading page when login

wedontkn0w, 2020-02-21:
### Actual behavior
Hello. Im stuck in loading dialog after submit login. After my first set up it works I connected succesful and in a few minutes I cannot login I stuck in loading dialog. I set  it up again and still I'm stuck with loading dialog. Did I miss something? Need help.

### Your environment
- Hardware : rasoberry pi 3
- Operating system: raspbian buster lite

pihomeserver, 2020-02-21:
You cleaned the issue template without providing requested information so no idea of what could be the problem

wedontkn0w, 2020-02-21:
Please tell me what are the possible problem. When encountering like this. Chilli, freeradius, mariadb service are online.

pihomeserver, 2020-03-06:
See https://github.com/pihomeserver/Kupiki-Hotspot-Script/issues/199#issuecomment-589628765

ardinusawan, 2020-05-01:
Hi @pihomeserver 
I have similar issue, here my Nginx access log

```sh
192.168.10.3 - - [01/May/2020:12:50:18 +0700] "GET /?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204 HTTP/1.1" 200 1319 "-" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /css/bootstrap.min.css HTTP/1.1" 304 0 "http://192.168.10.1/?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /css/portal.css HTTP/1.1" 304 0 "http://192.168.10.1/?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /js/jquery.min.js HTTP/1.1" 304 0 "http://192.168.10.1/?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /js/ChilliLibrary.js HTTP/1.1" 304 0 "http://192.168.10.1/?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /css/spinner.css HTTP/1.1" 304 0 "http://192.168.10.1/?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /js/portal.js HTTP/1.1" 304 0 "http://192.168.10.1/?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /images/background.jpg HTTP/1.1" 304 0 "http://192.168.10.1/css/portal.css" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /js/configuration.json HTTP/1.1" 304 0 "http://192.168.10.1/?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:19 +0700] "GET /favicon.ico HTTP/1.1" 404 199 "http://192.168.10.1/?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fconnect.rom.miui.com%2fgenerate_204" "Mozilla/5.0 (Linux; Android 10; MI 8 Build/QKQ1.190828.002; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/81.0.4044.117 Mobile Safari/537.36"
192.168.10.3 - - [01/May/2020:12:50:23 +0700] "GET /?res=notyet&uamip=192.168.10.1&uamport=3990&challenge=9705c4379fbb7f7f778d55301940b70b&called=B8-27-EB-B0-7C-0A&mac=94-87-E0-0A-43-05&ip=192.168.10.3&ssid=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fresolver.msg.global.xiaomi.net%2fgslb%2f%3fver%3d4.0%26type%3dwifi%26uuid%3d0%26list%3dapp.chat.globa [... cut ...] id=raspberrypi&nasid=KUPIKI&sessionid=158831221800000003&userurl=http%3a%2f%2fresolver.msg.global.xiaomi.net%2fgslb%2f%3fver%3d4.0%26type%3dwifi%26uuid%3d0%26list%3dapp.chat.global.xiaomi.net%252Cresolver.msg.global.xiaomi.net%26countrycode%3dID%26sdkver%3d39%26osver%3d29%26os%3dMI%25208%253AV11.0.4.0.QEAMIXM%26mi%3d3%26key%3d56698a8dc77c4ed1da6601849ef36c63 HTTP/1.1" 200 1319 "-" "Dalvik/2.1.0 (Linux; U; Android 10; MI 8 MIUI/V11.0.4.0.QEAMIXM)"
```

No error on Nginx error.log

Please advice 🙏

ardinusawan, 2020-05-01:
And here log from daloRadius
```sh
Fri May 1 13:07:06 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 13:06:56 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 13:06:46 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 13:06:36 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 13:06:26 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
```

Eventhought username and password is correct, but why it's say incorrect ya?

ardinusawan, 2020-05-01:
Last 50 lines log daloRadius
```Fri May 1 13:07:06 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 13:06:56 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 13:06:46 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 13:06:36 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 13:06:26 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:51:11 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:51:01 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:50:51 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:50:41 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:50:31 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:46:32 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:46:22 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:46:12 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:46:02 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:45:52 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:45:27 2020 : Info: Dropping packet without response because of error: Received Accounting-Request packet from client 127.0.0.1 with invalid Request Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:45:17 2020 : Info: Dropping packet without response because of error: Received Accounting-Request packet from client 127.0.0.1 with invalid Request Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:45:00 2020 : Info: Dropping packet without response because of error: Received Accounting-Request packet from client 127.0.0.1 with invalid Request Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:44:54 2020 : Info: Ready to process requests 
Fri May 1 12:44:54 2020 : Info: Loaded virtual server inner-tunnel 
Fri May 1 12:44:54 2020 : Info: # Skipping contents of 'if' as it is always 'false' -- /etc/freeradius/3.0/sites-enabled/inner-tunnel:331 
Fri May 1 12:44:54 2020 : Info: Loaded virtual server default 
Fri May 1 12:44:54 2020 : Warning: Ignoring "ldap" (see raddb/mods-available/README.rst) 
Fri May 1 12:44:54 2020 : Info: Loaded virtual server coa 
Fri May 1 12:44:54 2020 : Info: Loaded virtual server 
Fri May 1 12:44:54 2020 : Info: rlm_sql (sql): Opening additional connection (4), 1 of 28 pending slots used 
Fri May 1 12:44:54 2020 : Info: rlm_sql (sql): Opening additional connection (3), 1 of 29 pending slots used 
Fri May 1 12:44:54 2020 : Info: rlm_sql (sql): Opening additional connection (2), 1 of 30 pending slots used 
Fri May 1 12:44:54 2020 : Info: rlm_sql (sql): Opening additional connection (1), 1 of 31 pending slots used 
Fri May 1 12:44:54 2020 : Info: rlm_sql (sql): Opening additional connection (0), 1 of 32 pending slots used 
Fri May 1 12:44:54 2020 : Info: rlm_sql (sql): Attempting to connect to database "radius" 
Fri May 1 12:44:54 2020 : Info: rlm_sql_mysql: libmysql version: 10.3.22 
Fri May 1 12:44:54 2020 : Warning: [/etc/freeradius/3.0/mods-config/attr_filter/access_reject]:11 Check item "FreeRADIUS-Response-Delay-USec" found in filter list for realm "DEFAULT". 
Fri May 1 12:44:54 2020 : Warning: [/etc/freeradius/3.0/mods-config/attr_filter/access_reject]:11 Check item "FreeRADIUS-Response-Delay" found in filter list for realm "DEFAULT". 
Fri May 1 12:44:54 2020 : Info: rlm_sql (sql): Driver rlm_sql_mysql (module rlm_sql_mysql) loaded and linked 
Fri May 1 12:44:53 2020 : Info: Debugger not attached 
Fri May 1 12:43:55 2020 : Info: rlm_sql (sql): Closing connection (0) 
Fri May 1 12:43:55 2020 : Info: rlm_sql (sql): Closing connection (1) 
Fri May 1 12:43:55 2020 : Info: rlm_sql (sql): Closing connection (2) 
Fri May 1 12:43:55 2020 : Info: rlm_sql (sql): Closing connection (3) 
[... cut ...]
Fri May 1 12:29:53 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:29:32 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
Fri May 1 12:29:22 2020 : Info: Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator! (Shared secret is incorrect.) 
```

pihomeserver, 2020-05-01:
How did you create the user ? With the cleartext password policy ?

ardinusawan, 2020-05-01:
> How did you create the user ? With the cleartext password policy ?

Yes correct. How it should?

pihomeserver, 2020-05-04:
Do you see the user in the database with its attributes ? At least to be sure that the creation worked. Else try to create one using the radclient

ardinusawan, 2020-05-08:
@pihomeserver On `<host>/mng-list-all.php`, It exist

ardinusawan, 2020-05-08:
So I follow [this tutorial](https://wiki.freeradius.org/guide/Basic-configuration-HOWTO)

I put this at the top of the file user on `/etc/freeradius/3.0/users` with value: `testing Cleartext-Password := "password"`

I disable free radius on systemctl, and then run it manually using debug mode
```sh
pi@raspberrypi:~ $ sudo /usr/sbin/freeradius -X
FreeRADIUS Version 3.0.17
Copyright (C) 1999-2017 The FreeRADIUS server project and contributors
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE
You may redistribute copies of FreeRADIUS under the terms of the
GNU General Public License
For more information about these matters, see the file named COPYRIGHT
Starting - reading configuration files ...
including dictionary file /usr/share/freeradius/dictionary
including dictionary file /usr/share/freeradius/dictionary.dhcp
including dictionary file /usr/share/freeradius/dictionary.vqp
including dictionary file /etc/freeradius/3.0/dictionary
including configuration file /etc/freeradius/3.0/radiusd.conf
including configuration file /etc/freeradius/3.0/proxy.conf
including configuration file /etc/freeradius/3.0/clients.conf
including files in directory /etc/freeradius/3.0/mods-enabled/
including configuration file /etc/freeradius/3.0/mods-enabled/ntlm_auth
including configuration file /etc/freeradius/3.0/mods-enabled/files
including configuration file /etc/freeradius/3.0/mods-enabled/soh
including configuration file /etc/freeradius/3.0/mods-enabled/attr_filter
including configuration file /etc/freeradius/3.0/mods-enabled/radutmp
including configuration file /etc/freeradius/3.0/mods-enabled/eap
including configuration file /etc/freeradius/3.0/mods-enabled/cache_eap
including configuration file /etc/freeradius/3.0/mods-enabled/always
including configuration file /etc/freeradius/3.0/mods-enabled/passwd
including configuration file /etc/freeradius/3.0/mods-enabled/realm
including configuration file /etc/freeradius/3.0/mods-enabled/exec
including configuration file /etc/freeradius/3.0/mods-enabled/mschap
including configuration file /etc/freeradius/3.0/mods-enabled/echo
including configuration file /etc/freeradius/3.0/mods-enabled/digest
including configuration file /etc/freeradius/3.0/mods-enabled/detail
including configuration file /etc/freeradius/3.0/mods-enabled/sql
including configuration file /etc/freeradius/3.0/mods-config/sql/main/mysql/queries.conf
including configuration file /etc/freeradius/3.0/mods-enabled/expr
including configuration file /etc/freeradius/3.0/mods-enabled/unpack
including configuration file /etc/freeradius/3.0/mods-enabled/dynamic_clients
including configuration file /etc/freeradius/3.0/mods-enabled/sqlcounter
including configuration file /etc/freeradius/3.0/mods-config/sql/counter/mysql/dailycounter.conf
including configuration file /etc/freeradius/3.0/mods-config/sql/counter/mysql/monthlycounter.conf
including configuration file /etc/freeradius/3.0/mods-config/sql/counter/mysql/noresetcounter.conf
including configuration file /etc/freeradius/3.0/mods-config/sql/counter/mysql/expire_on_login.conf
including configuration file /etc/freeradius/3.0/mods-enabled/unix
including configuration file /etc/freeradius/3.0/mods-enabled/linelog
including configuration file /etc/freeradius/3.0/mods-enabled/sradutmp
including configuration file /etc/freeradius/3.0/mods-enabled/logintime
including configuration file /etc/freeradius/3.0/mods-enabled/replicate
including configuration file /etc/freeradius/3.0/mods-enabled/detail.log
including configuration file /etc/freeradius/3.0/mods-enabled/expiration
including configuration file /etc/freeradius/3.0/mods-enabled/preprocess
including configuration file /etc/freeradius/3.0/mods-enabled/chap
including configuration file /etc/freeradius/3.0/mods-enabled/utf8
including configuration file /etc/freeradius/3.0/mods-enabled/pap
including files in directory /etc/freeradius/3.0/policy.d/
including configuration file /etc/freeradius/3.0/policy.d/eap
including configuration file /etc/freeradius/3.0/policy.d/canonicalization
including configuration file /etc/freeradius/3.0/policy.d/debug
including configuration file /etc/freeradius/3.0/policy.d/accounting
including configuration file /etc/freeradius/3.0/policy.d/moonshot-targeted-ids
including configuration file /etc/freeradius/3.0/policy.d/cui
including configuration file /etc/freeradius/3.0/policy.d/filter
including configuration file /etc/freeradius/3.0/policy.d/abfab-tr
including configuration file /etc/freeradius/3.0/policy.d/control
including configuration file /etc/freeradius/3.0/policy.d/dhcp
[... 1605 lines cut ...]
Waking up in 4.9 seconds.
(0) Cleaning up request packet ID 61 with timestamp +96
Ready to process requests
```

And when I try to login using username: `testing`, password: `password` I got this log
```sh
(1) Received Access-Request Id 66 from 127.0.0.1:56358 to 127.0.0.1:1812 length 307
Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator!  (Shared secret is incorrect.)
Waking up in 0.3 seconds.
(1) Cleaning up request packet ID 66 with timestamp +442
Ready to process requests
```
So I assume it success.

But when I try to login with same username dan password on other device, I got
```sh
(1) Received Access-Request Id 66 from 127.0.0.1:56358 to 127.0.0.1:1812 length 307
Dropping packet without response because of error: Received packet from 127.0.0.1 with invalid Message-Authenticator!  (Shared secret is incorrect.)
Waking up in 0.3 seconds.
(1) Cleaning up request packet ID 66 with timestamp +442
Ready to process requests
```

Please help @pihomeserver  🙏

pihomeserver, 2020-05-08:
Other device than where Freeradius is installed ? If so that's normal as you send your request to localhost that is the current devide not the one with Freeradius

ardinusawan, 2020-05-08:
Oh ok. 

Btw I query the user on DB, and the value is there (same as freeradius show on website)
```sh
MariaDB [radius]> select * from userinfo;
+----+----------+-----------+----------+-------+------------+---------+-----------+-----------+-------------+---------+------+-------+---------+------+-------+----------------+---------------------+-------------------+---------------------+---------------+------------+----------+
| id | username | firstname | lastname | email | department | company | workphone | homephone | mobilephone | address | city | state | country | zip  | notes | changeuserinfo | portalloginpassword | enableportallogin | creationdate        | creationby    | updatedate | updateby |
+----+----------+-----------+----------+-------+------------+---------+-----------+-----------+-------------+---------+------+-------+---------+------+-------+----------------+---------------------+-------------------+---------------------+---------------+------------+----------+
|  1 | komang   |           |          |       |            |         |           |           |             |         |      |       |         |      |       | 0              |                     |                 0 | 2020-05-01 12:25:13 | administrator | NULL       | NULL     |
|  2 | wawan    |           |          |       |            |         |           |           |             |         |      |       |         |      |       | 0              |                     |                 0 | 2020-05-01 18:46:39 | administrator | NULL       | NULL     |
|  3 | test     |           |          |       |            |         |           |           |             |         |      |       |         |      |       | 0              |                     |                 0 | 2020-05-08 20:13:31 | administrator | NULL       | NULL     |
+----+----------+-----------+----------+-------+------------+---------+-----------+-----------+-------------+---------+------+-------+---------+------+-------+----------------+---------------------+-------------------+---------------------+---------------+------------+----------+
3 rows in set (0.001 sec)
```
Why ya?

Oh and can you give me the command how to create user with radclient? @pihomeserver

pihomeserver, 2020-05-09:
I suggest to use Google ... https://wiki.freeradius.org/config/Radclient#command-line