pihomeserver/Kupiki-Hotspot-Script #200: Accessing thorugh ethernet gives 502 Bad Gateway

ccvelandres, 2020-02-26:
Fresh Install on Raspbian Buster Lite 2020-02-13

### Actual behavior

The hostpot works with the portal correctly. But when I try to access through the ethernet interface, nginx gives me a error 502 Bad Gateway.

### Expected behavior

{Please write here}

#### Configuration parameters (if changed in the script)

I didn't change anything from the script

#### Logs

Log file is attached

### Your environment

- Hardware : RPI 4
- Operating system: Raspbian Buster Lite

### Additional comments

[kupiki_hotspot.log](https://github.com/pihomeserver/Kupiki-Hotspot-Script/files/4255348/kupiki_hotspot.log)

pihomeserver, 2020-02-26:
Are you sure about the network configuration in particular the IP address of the interfaces

ccvelandres, 2020-02-27:
Yes, I did another fresh install this time I changed the network for the hostpot to 172.16.10.0

eth0: inet 192.168.137.209 netmask 255.255.255.0 broadcast 192.168.137.255
tun0: inet 172.16.10.1 netmask 255.255.255.0 broadcast 172.16.10.1

Again, the hotspot portal works fine but when I access through eth0 (192.168.137.209) it gives me another 502 Bad Gateway. 
The network configuration must be right since I can access the pi through the ethernet.

I think the problem might be in nginx configuration? But I'm not familiar with nginx

ccvelandres, 2020-02-27:
Here's a line from /var/log/nginx/error.log

2020/02/27 11:38:44 [crit] 17870#17870: *107 connect() to unix:/var/run/php/php7.0-fpm.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.137.1, server: _, request: "GET /daloradius/login.php HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock:", host: "192.168.137.209"

It seems like it can't find login.php? I checked the daloradius directory but the login.php file is there. 

[access.log](https://github.com/pihomeserver/Kupiki-Hotspot-Script/files/4261460/access.log)
[error.log](https://github.com/pihomeserver/Kupiki-Hotspot-Script/files/4261461/error.log)

ccvelandres, 2020-02-27:
Found the problem.

in /etc/nginx/sites-available/default
```
            location ~ \.php$ {
                include snippets/fastcgi-php.conf;
                fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
            }
```
The line:
fastcgi_pass unix:/var/run/php/php7.0-fpm.sock; 
should be
fastcgi_pass unix:/var/run/php/php7.**3**-fpm.sock;

closed 2020-02-27 by pihomeserver