pihomeserver/Kupiki-Hotspot-Script #205: Issue during installation

LoloPicolo, 2020-07-18:
### Actual behavior

on a fresh installation of Raspberry Pi OS (32-bit) Lite
Version: May 2020
Release date: 2020-05-27
Kernel version: 4.19
i execute installation script with default config values.

i have an error

::: Activating the LAN interface
RTNETLINK answers: Operation not possible due to RF-kill
ifup: failed to bring up wlan0
::: 
::: Something went wrong with the last command. Please check the log file
::: 

### Expected behavior

Standard installation

#### Configuration parameters (if changed in the script)

no changes in script

#### Logs

Listening on LPF/eth0/b8:27:eb:af:7d:eb
Sending on   LPF/eth0/b8:27:eb:af:7d:eb
Sending on   Socket/fallback
Created duid "\000\001\000\001&\243g\016\270'\353\257}\353".
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 3
DHCPOFFER of 192.168.20.72 from 192.168.20.254
DHCPREQUEST for 192.168.20.72 on eth0 to 255.255.255.255 port 67
DHCPACK of 192.168.20.72 from 192.168.20.254
RTNETLINK answers: File exists
bound to 192.168.20.72 -- renewal in 20876 seconds.
::: Activating the LAN interface
RTNETLINK answers: Operation not possible due to RF-kill
ifup: failed to bring up wlan0
::: 
::: Something went wrong with the last command. Please check the log file
::: 

### Your environment

Raspberry 3B
Raspberry Pi OS (32-bit) Lite Version: May 2020 Release date: 2020-05-27 Kernel version: 4.19

### Additional comments

[kupiki_hotspot.log](https://github.com/pihomeserver/Kupiki-Hotspot-Script/files/4941729/kupiki_hotspot.log)

pihomeserver, 2020-07-30:
Can you try to restart the script once ?

modhash, 2020-11-24:
I have the same issue and I tried to rerun the script multiple times it did not help

pihomeserver, 2020-11-25:
Try to look at that feed : https://www.raspberrypi.org/forums/viewtopic.php?t=146198

modhash, 2020-11-25:
it did not help

Belluxx, 2021-08-09:
Had the same problem on the Raspberry Pi 4 and i solved it:
`sudo echo "blacklist hp_wmi" > /etc/modprobe.d/hp.conf`
`sudo reboot`
`sudo rfkill unblock all`

Credit: https://bbs.archlinux.org/viewtopic.php?pid=1324810#p1324810

pihomeserver, 2021-08-15:
Hi
Thank you for the reply and share of the tip
Can you submit a PR to update the script according to your solution ?