pihomeserver/Kupiki-Hotspot-Script #208: Run mitmproxy on Kupiki Hotspot

lt3stus3el, 2021-06-17:
I have installed kupiki hotspot from the script and it works without problem. I windows if I can run mitmproxy on it with transparent mode? If so, which port I should forward packets and from which interface?

I have tried to forward packets from tun0 (which are detisnated to port 80 and 443) to port 8080 (in order to mitmproxy can work) but two things happened:

1. When I forward port 80, the login page of the Kupiki hotspot gone unavailable.
2. When I forward port 443, the client can not load any web pages.

I am using /sbin/iptables for the changes. And I'm following the instructions from here:
https://groups.google.com/g/mitmproxy/c/4htH_ujmOdQ

Thanks

pihomeserver, 2021-06-17:
Sorry i don't really know how to achieve that. What i know is that after a stop/start, Coova updates iptables

You can have a look here for tinyproxy
https://www.radiusdesk.com/old_wiki/technical_discussions/tinyproxy_ubuntu

lt3stus3el, 2021-06-19:
Thanks for the answer. I will check it.