pihomeserver/Kupiki-Hotspot-Script #210: Can't Install Kupiki-Hotspot-Script

Mark-Alfred, 2022-09-17:
### Actual behavior

Can't get past the 
**: : : Updating /etc/hosts**

### Expected behavior

I expect that the installation goes according to plan.

#### Configuration parameters (if changed in the script)

HOTSPOT_NAME="Vianney's Free Wi-Fi"
WIFI_COUNTRY_CODE="PH"

#### Logs

```
: : : Your system is up to date! Continuing with Pi-Hotspot installation...
: : : Disable integrated Bluetooth support (After next reboot)
: : : Remove old configuration of Coova Chilli
: : : Remove old configuration of haserl
: : : Remove old configuration of hostapd
: : : Checking for tun module
: : : Checking if LAN interface already exists
: : : Adding MariaDB password
: : : Adding MariaDB password (confirmation)
: : : Getting WAN IP of the Raspberry Pi (for dalodarius access)
: : : Updating the system hostname to Vianney's Free Wi-Fi
: : : Adding Avahi dependencies
: : : Updating /etc/hosts
./pihotspot.sh: eval: line 156: unexpected EOF while looking for matching ` ' '
./pihotspot.sh: eval: line 157: syntax error: unexpected end of file
: : : Correct configuration for Collectd daemon
sed: can't read /etc/collectd/collectd.conf: No such file or directory
: : : 
: : : Something went wrong with the last command. Please check the log file
: : : 
root@raspberrypi:/home/pi/Kupiki-Hotspot-Script#
```

### Your environment

- Hardware : Raspberry Pi 3B+
- Operating system: Raspberry Pi OS Lite (64 BIT)

### Additional comments

Please help. I have tried installing this several times. I have tried to install it on the Raspberry Pi OS Full (64 BIT) thrice, and I have tried to install this on the Raspberry Pi OS Lite (64 BIT), I have no clue what I'm doing wrong as the only thing that I edited was the configuration parameters for the Hotspot Name and WIFI Country Code. I'm hoping someone can help me on this. Thank you so much!

chivarpilones, 2023-04-21:
do a sudo apt install collectd -y

yelooooo422, 2023-08-19:
Hi, I already installed collectd -y and went back to installation process of hotspot,sh. However, I met again another error saying 

::: Install pika module
./pihotspot.sh: line 508: pip: command not found
:::
::: Something went wrong with the last command. Please check the log file

Can someone help me with this?