pihomeserver/Kupiki-Hotspot-Script #211: Checking freeradius configuration

chivarpilones, 2023-04-21:
### Actual behavior

::: Checking freeradius configuration
::: 
::: Something went wrong with the last command. Please check the log file

### Expected behavior

{Please write here}

#### Configuration parameters (if changed in the script)

#### Logs

```
::: 
::: Log file created : /var/log/kupiki_hotspot.log
::: Use command 'tail -f /var/log/kupiki_hotspot.log' in a new console to get installation details
::: Updating APT config to force IPv4
Acquire::ForceIPv4 "true";
::: Configuring localepurge
::: Disable integrated Bluetooth support (After next reboot)
::: Remove old configuration of Coova Chilli
dpkg: warning: ignoring request to remove coova-chilli which isn't installed
::: Remove old configuration of haserl
dpkg: warning: ignoring request to remove haserl which isn't installed
::: Remove old configuration of hostapd
dpkg: warning: ignoring request to remove hostapd which isn't installed
::: Checking for tun module
tun                    61440  0
::: Checking if LAN interface already exists
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
::: Adding MariaDb password
::: Adding MariaDb password (confirmation)
::: Getting WAN IP of the Raspberry Pi (for daloradius access)
::: Updating the system hostname to kupikihotspot
::: Adding Avahi dependencies
::: Updating /etc/hosts
::: Correct configuration for Collectd daemon
::: Install pika module
::: Removing any previous sources of CoovaChilli project
::: Cloning CoovaChilli project
Cloning into 'coova-chilli'...
::: Removing any previous sources of Kupiki Admin
::: Cloning Kupiki Admin web UI project
Cloning into 'webui'...
::: Download Kupiki SQL Counters
--2023-04-21 23:10:26--  https://raw.githubusercontent.com/pihomeserver/Kupiki-Hotspot-Script/master/updates/sqlcounter
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.109.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9037 (8.8K) [text/plain]
Saving to: ‘sqlcounter’

     0K ........                                              100% 7.53M=0.001s

2023-04-21 23:10:28 (7.53 MB/s) - ‘sqlcounter’ saved [9037/9037]

::: Cloning daloradius project
Cloning into 'daloradius'...
Updating files: 100% (1654/1654), done.
::: Cloning Pi Hotspot portal project
Cloning into 'portal'...
::: Cloning Pi Hotspot portal backend project
Cloning into 'kupiki-portal-backend'...
::: Starting MySql service
::: Update interface configuration (eth0)
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp
::: Update interface configuration (wlan0)
auto wlan0
allow-hotplug wlan0
iface wlan0 inet static
::: Update wifi configuration to add country code
country=PH
::: Activating the WAN interface
ifup: interface eth0 already configured
::: Activating the LAN interface
ifup: interface wlan0 already configured
::: Stopping fprobe service
::: Stopping nfdump service
::: Updating fprobe configuration
::: Updating nfdump configuration
::: Updating nfdump service configuration (init.d)
::: Updating nfdump service configuration (systemd)
::: Set cleaning options for Netflow data
::: Adding logs cleaning every day at 1am (system time)
::: Adding Netflow stats update in the crontab
00 01 * * * root /usr/bin/nfexpire -r /var/cache/nfdump
::: Stopping freeradius service to update the configuration
::: Creating freeradius database
::: Installing freeradius schema
::: Adding setup data
::: Updating freeradius configuration - Activate SQL support
::: Configuration of the Freeradius SQL driver
::: Change dialect of the Freeradius SQL driver to mysql
::: Configuration of the Freeradius SQL connection
::: Updating freeradius configuration - Activate SQL counters
::: Bug fix for SQL dialect once SQL Counters are activated
::: Update of Freeradius secret key
::: Updating inner-tunnel configuration (1)
::: Updating inner-tunnel configuration (2)
::: Updating freeradius default configuration (1)
::: Updating freeradius default configuration (2)
::: Activating COA support in Freeradius
::: Adding CoovaChilli counters (limit bandwidth)
::: Updating file access rights
::: Activating counters
::: Checking freeradius configuration
::: 
::: Something went wrong with the last command. Please check the log file
:::
### Your environment

- Hardware :Raspberry Pi 3 Model B Plus Rev 1.3
- Operating system: Debian GNU/Linux 11 (bullseye) / Architecture: aarch64
- Whois: SPACEX-AP Address: 1 Rocket Rd, Hawthorne CA 90250 1 Rocket Rd Country: US

Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG    202    0        0 eth0
192.168.1.0     0.0.0.0         255.255.255.0   U     202    0        0 eth0
192.168.10.0    0.0.0.0         255.255.255.0   U     0      0        0 wlan0

### Additional comments

sudo apt update && sudo apt upgrade
sudo apt install collectd
sudo apt install mariadb-server
sudo apt install freeradius -y
cd /usr/src/coova-chilli && dpkg-buildpackage -us -uc
apt install debhelper -y
apt install gengetopt libtool automake libssl-dev libjson-c-dev

Could not link driver rlm_sql_my
[freeradius_errorlog.txt](https://github.com/pihomeserver/Kupiki-Hotspot-Script/files/11297338/freeradius_errorlog.txt)
sql: /usr/lib/freeradius/rlm_sql_mysql.so: cannot open shared object file: No such file or directory
Make sure it (and all its dependent libraries!) are in the search path of your system's ld
/etc/freeradius/3.0/mods-enabled/sql[27]: Instantiation failed for module "sql"

chivarpilones, 2023-04-21:
https://packages.debian.org/bullseye/arm64/freeradius-mysql/filelist

rlm_sql (sql): Driver rlm_sql_mysql (module rlm_sql_mysql) loaded and linked
Creating attribute SQL-Group
  # Loaded module rlm_files
  # Loading module "files" from file /etc/freeradius/3.0/mods-enabled/files
  files {
        filename = "/etc/freeradius/3.0/mods-config/files/authorize"
        acctusersfile = "/etc/freeradius/3.0/mods-config/files/accounting"
        preproxy_usersfile = "/etc/freeradius/3.0/mods-config/files/pre-proxy"
  }
  # Loaded module rlm_cache

###### now im having probs with #####
# Instantiating module "sql" from file /etc/freeradius/3.0/mods-enabled/sql
rlm_sql_mysql: libmysql version: 10.5.18
   mysql {
    tls {
        ca_file = "/etc/ssl/certs/my_ca.crt"
Unable to check file "/etc/ssl/certs/my_ca.crt": No such file or directory
/etc/freeradius/3.0/mods-enabled/sql[90]: Failed parsing configuration item "ca_file"
/etc/freeradius/3.0/mods-enabled/sql[27]: Instantiation failed for module "sql"
root@kupikihotspot:/home/chivar# NTPRadPing
bash: NTPRadPing: command not found
root@kupikihotspot:/home/chivar# mysql -u root -p
Enter password:

pihomeserver, 2023-08-19:
Password is defined at the beginning of the script